kashimAstro/ofxGPIO #2: PI2

antimodular, 2015-11-09:


kashimAstro, 2015-11-10:
@antimodular I'm working on porting all the addons under OF0.9
when they are short on ofxFurry I deal ofxGPIO

good day

grayfuse, 2016-02-29:
Any movement on updating this?

kashimAstro, 2016-03-01:
@grayfuse I'm building the version 2 dell'addons with new examples and new more practical functions, unfortunately the work keeps me a little busy these days, I predict that within a week I throw out a beta.

new future:
https://www.youtube.com/watch?v=XuHB4FBesbY

I keep you updated..

grayfuse, 2016-03-03:
Awesome, looking forward to seeing it, thanks for sharing your progress!

kashimAstro, 2016-03-14:
hello guys @antimodular  @grayfuse,
I wanted to tell you I made a first update of ofxGPIO.

I added a bit of functionality and some new examples,
as soon as I have time I prepare a detailed wiki.

good day

closed 2017-12-28 by kashimAstro