kashimAstro/ofxGPIO #3: Simulate GPIO

madc, 2016-03-29:
I'm currently developing an application using this addon and it an the RPI it works fine. But as soon, as I add the addon, I can't compile my app on my notebook (Arch) anymore, which makes sense, as it has no GPIO pins..

I didn't dig to deep into your code yet, but would it make sense to simulate the pins, if not available?
I was thinking similar to [this solution](https://github.com/bobvann/GPIOSim) for the python implementation.

Not sure, if it really makes sense, just throwing the idea out there.

kashimAstro, 2016-03-29:
HI @madc 

This addons can be compiled under the ARM architecture, if you want to compile this addons on propio PC with different architecture must create a cross compiling environment.

Look here to cross compile openFrameworks guides:
http://openframeworks.cc/setup/raspberrypi/raspberry-pi-cross-compiling-guide/
https://forum.openframeworks.cc/t/cross-compiler-for-of-0-9-0-jessie-arm6-rpi1/21336

the speech create a GPIO simulator looks really interesting, I thank you for the inspiration, just am free from work study the issue in depth.

I keep you updated.

good day

madc, 2016-03-31:
Thanks for the links, I'm already familiar with them, but have a somewhat unusual setup (Arch Linux ARM on the PI for example) and try to avoid cross-compiling for now.

I just stumbled across [CppGPIO](https://github.com/JoachimSchurig/CppGPIO), which seems to work quite well and also has the possibility to [simulate GPIOs](https://github.com/JoachimSchurig/CppGPIO/blob/master/include/cppgpio/gpio.hpp#L118). Have a look.

In my app, i use it like this:
```
#ifndef __arm__
GPIO::GPIOBase::simulation(true);
#endif
```

kashimAstro, 2016-03-31:
Thank you @madc  for this cue,
Last night, I started a small implementation for a simulator I/O GPIO, as you have suggested.

I thought of it a small graphical interface with the two versions of GPIO [A / B] and [A + / B + / 2 / 3] with the ability to enable and disable I/O pin with a click...

and still a little hazy.. all I keep you updated.

if you have other suggestions I'm here to listen!!!

good day

kashimAstro, 2016-03-31:
[image]

kashimAstro, 2016-03-31:
It might be interesting to the management of protocols such as: I2C and SPI, for the simulator....

for i2c you can think of something like this:
https://labjack.com/content/i2c-simulator

noOblsa1botl, 2017-10-26:
https://yadi.sk/i/FEpP1-3C3P8HYs Библиотека для Джойстика 

https://yadi.sk/i/QdtruzHL3P8HYw Библиотека для ЛСД дисплея