kashimAstro/ofxGPIO #4: SPI not working in of093 and RPi2

urbanpixellab, 2016-06-13:
Hi tested the spi example with the connection to the adc3008 shown in the readme but i always get the error "Problem transmitting spi data..ioctl". I tried the button example and this is working fine.

Thanks in advance.

kashimAstro, 2016-06-13:
Hi, I will make a couple of quick questions, to understand the problem...

1) you have enabled SPI? There are many ways to do it, simple mode:
  raspi-config -> Advanced Option -> SPI

2) to have a confirmation on the operation of SPI do:
lsmod | grep bcm

3) you can make a photo at the cable connection?

4) since the last update and 'added a SPI2 class that I have not tested adc3008, you are using SPI or SPI2?

thanks for the tip :)

urbanpixellab, 2016-06-13:
1. spi is already enabled

2. 
snd_bcm2835            20511  1 
snd_pcm                75698  1 snd_bcm2835
snd                    51844  5 snd_bcm2835,snd_timer,snd_pcm
bcm2835_wdt             3225  0 
bcm2835_gpiomem         3040  0 
spi_bcm2835             7286  0 

4. i am using spi, not spi2

3 pictures are in a few minutes

urbanpixellab, 2016-06-13:
[image]

kashimAstro, 2016-06-13:
you can publish your program setup that contain the "new = SPI()" ?

urbanpixellab, 2016-06-13:
#include "ofMain.h"
#include "ofAppNoWindow.h"
#include "ofxGPIO.h"

class noWin : public ofBaseApp{
        public:
		SPI * spi;
	        int i;
	        int a2dVal;
	        int a2dChannel;
	        unsigned char data[3];

		void setup(){
		        i = 20;
		        a2dVal = 0;
		        a2dChannel = 0;
		        spi = new SPI("/dev/spidev0.0", SPI_MODE_0, 1000000, 8);
		}
		
		void update(){
			data[0] = 1;
		        data[1] = 0b10000000 |( ((a2dChannel & 7) << 4));
		        data[2] = 0;

		        spi->spiWriteRead(data, sizeof(data) );
 		        a2dVal = 0;
                	a2dVal = (data[1]<< 8) & 0b1100000000;
	                a2dVal |=  (data[2] & 0xff);
        		sleep(2);
		        ofLog()<<"Result: " << a2dVal;
		}
};

int main( ){
        ofAppNoWindow window;
	ofSetupOpenGL(&window, 0,0, OF_WINDOW);
	ofRunApp( new noWin() );
}

urbanpixellab, 2016-06-13:
This is your spi example, which i am using.

kashimAstro, 2016-06-13:
you've tried to use interface: /dev/spidev0.1  ?

urbanpixellab, 2016-06-13:
i tried now
spi = new SPI("/dev/spidev0.1", SPI_MODE_0, 1000000, 8);

but the same error.

kashimAstro, 2016-06-14:
I'm trying MCP3008 in the laboratory, short update you

kashimAstro, 2016-06-14:
ok the problem exists, but appears to be closely tied to 9.x of openFrameworks paired with a combination of reading from SPI.

In fact, if you try to run this example outside openFrameworks or with a version of openFrameworks < 9.x running fine.

I am attaching an example in C ++ to use without openFrameworks.
extract the zip file in the /home/pi/
cd mcp3008/
make
./main

[mcp3008.zip](https://github.com/kashimAstro/ofxGPIO/files/314321/mcp3008.zip)

let me know if this example outside openFrameworks works for you.
in a while I try to figure out the problem, I leave this open.

urbanpixellab, 2016-06-14:
wow, thank you for this example, this is working.

xumo, 2016-08-02:
Have you solved this issue yet?

kashimAstro, 2016-08-04:
hello @xumo , I'm sorry I fix this in September I am now out for work.

xumo, 2016-08-04:
Do you have any hints, I need deliver earlier? I'l try with OF 8.4, but cannot seem to find the release tar in the main site. 
Thanks anyway, this is an awesome addons.

kashimAstro, 2016-08-05:
[kit_mcp.zip](https://github.com/kashimAstro/ofxGPIO/files/403764/kit_mcp.zip)

Hi @xumo 

a simple solution without changing the version is to use the demon that you find in the zip writing in a socket are also an example openFrameworks to read socket.

Good day

xumo, 2016-08-12:
Yei, thanks a lot.

kashimAstro, 2016-08-30:
Hi guys @urbanpixellab @xumo, 

I think I've solved the problem with SPI.

good day

closed 2016-08-30 by kashimAstro

xumo, 2016-09-01:
That's great. I'll test it as soon as I can.