kashimAstro/ofxGPIO #5: Issue with a raspberry 3 B+

Taprik, 2016-06-13:
Hello,

I've tried some of the examples of the ofxgpio and I have some issues with the simple ones.
I'm using a raspberry 3 B+ with of 0.9.3.
It compiles very well.
The simple button example runs perfectly.
But None of the simple led or the relay example. 
When I check the tension between the gpio pin and the ground it stays at 0V.

Actually I managed to blink a led with command line I found at
http://www.blaess.fr/christophe/2012/11/26/les-gpio-du-raspberry-pi/
Here is the result in the terminal
[image]
I'm a beginner, but it seems to be the same code as in the addon.
Did I miss something ?

kashimAstro, 2016-06-14:
I just did a test and it looks like it works.
make sure the pins are free, always making unexport_gpio exit the program.

launch the program as superuser with: sudo ./bin/example-simple-led

you can check whether resources are released by controlling the folder:
ls /sys/class/gpio/

good day

Taprik, 2016-06-14:
Bonjour,
    
    Merci d'utiliser l'adresse suivante : contact@smart-machines.net
      
      Bien cordialement
      Patrick Suchet
      Smart Machines

kashimAstro, 2016-06-14:
usually I do not recommend to change the permissions on the folder sys, for his ties to kernel mode.

but it can be done easily if you know your system.
even by simply using:

chmod 777 namedeviceorfile
or
chown pi: pi namedeviceorfile

good day

closed 2016-06-14 by kashimAstro