kashimAstro/ofxGPIO #6: [ error ] Problem transmitting spi data..ioctl

dammianhr, 2016-09-13:
hi!
i have this problem when try to compiling with of OF 0.9.3 on raspberry pi 1 b+
i try to fix this with add memset(&spi,0,sizeof spi); after struct spi_ioc_transfer spi[length] defined
and and/or add memset(&spi[i], 0, sizeof (spi[i])); inside the for loop. but this not work.
any idea?

sorry for my inglish.
thanks for share!

kashimAstro, 2016-09-13:
Hi @dammianhr,

may I ask what are you using integrated? It needs for ADC MCP3008 / MCP3004 for analog input?
If so, you should download the entire addons, there is now a MCP class for ADC management.
if it is other integrated,  give me the datasheet, SPI currently the situation is a little confused, there are two SPI classes and I have to decide to eliminate a short it.

I can help out.

good day

dammianhr, 2016-09-13:
hi! @kashimAstro 
I'm trying to run this module:
http://www.hoperf.com/rf_transceiver/lora/RFM95W.html
i have the same error with the module plugged and unplugged

kashimAstro, 2016-09-13:
1) You have enabled SPI? You can be done easily with the command:
`sudo raspi-config -> Advanced Options -> SPI`

2) After enabling SPI verify loading the module with:
`lsmod | grep spi`
and the creation of the device with the command:
`ls /dev/spi*`

3) controls which PIN SPI you used, raspberry has two integrated SPI interfaces:
```
P1-24    CE0  /dev/spidev0.0
P1-26    CE1  /dev/spidev0.1

```
according to the PIN used, adjust the first parameter of: spi.setup("?",mode,speed,bits);
I have not read the datasheet, checks the SPI speed supported by the module.

hello
Dario

dammianhr, 2016-09-13:
i get this

lsmod | grep spi
spi_bcm2835             8032  0 
ls /dev/spi*
/dev/spidev0.0  /dev/spidev0.1

i use /dev/spidev0.0.

i test it in python and work fine!

kashimAstro, 2016-09-14:
Hi @dammianhr,

pass me the python code.

dammianhr, 2016-09-14:
import spidev

spi = spidev.SpiDev()
spi.open(0,0)

to_send = [(0x01 | 0x80),(0x00 | 0x80)]
spi.writebytes(to_send)

to_read = [0x01]
spi.writebytes(to_read)

recived = spi.readbytes(0)

if recived == [(0x00 | 0x80)]:
        print "lora mode on!\n"

print recived

kashimAstro, 2016-09-15:
Hi @dammianhr,
try using the SPI2 class, right now I can not verify this code work, try to see if it works.

```
#include "ofMain.h"
#include "ofxGPIO.h"

class ofApp : public ofBaseApp{

        public:
        SPI2 spi;

        void setup(){
                int channel = 0;
                spi.setup(channel,1000000);
                char data[2];
                data[0] = (0x01 | 0x80);
                data[1] = (0x00 | 0x80);
                char *p = data;
                if( spi.readWrite(channel,p,sizeof(data)) )
                {
                        ofLog()<<"Send Request OK!";
                }
        }
};

int main()
{
        ofSetupOpenGL(1024,768, OF_WINDOW);
        ofRunApp( new ofApp() );
}
```

Ciao

dammianhr, 2016-09-22:
hi @kashimAstro !
thanks for your help,

I did several tests and found that my problem was a rare performance of the CS pin .
In python i defined a new pin CS and works well , but not so with ofxGPIO
python code:
`import spidev
import time
import RPi.GPIO as GPIO

GPIO.setmode(GPIO.BCM)
GPIO.setup(22, GPIO.OUT)
GPIO.setup(24, GPIO.OUT)

GPIO.output(24,True)

GPIO.output(22,False)
time.sleep(0.5)
GPIO.output(22, True)

spi = spidev.SpiDev()
spi.open(0,0)

spi.bits_per_word = 8
spi.mode = 0
spi.max_speed_hz = 3000000

to_send = [(0x01 | 0x80), (0x00 | 0x80)]

GPIO.output(24,False)
spi.writebytes(to_send)
GPIO.output(24,True)
time.sleep(0.01)

GPIO.output(24,False)
to_read = [0x01]
spi.writebytes(to_read)
recived = spi.readbytes(0)
GPIO.output(24,True)

if recived == [(0x00 | 0x80)]:
        print "lora mode on!\n"

print "OP_MODE: ",BytesToHex(recived) , "bin: ", recived`

kashimAstro, 2016-09-23:
Hi @dammianhr,

Maybe I understand the problem, you are using the wrong GPIO numbering, python libraries maintain an internal map of the pin that ofxGPIO do not.

for the execution of the ofxGPIO pins you have to use internal numbers, look at the following image:

[image]

check patch: "new GPIO("number pin")"

```
#include "ofMain.h"
#include "ofxGPIO.h"

#define RH_SPI_WRITE_MASK            0x80
#define RH_RF95_REG_01_OP_MODE       0x01
#define RH_RF95_MODE_SLEEP           0x00
#define RH_RF95_LONG_RANGE_MODE      0x80

class ofApp : public ofBaseApp{
        public:
        GPIO *cs;
        GPIO *rst;
        SPI2 spi;

        void setup(){
                int ch = 0;

                cs  = new GPIO("18"); //GPIO24
                cs->export_gpio();
                cs->setdir_gpio("out");
                cs->setval_gpio("1");

                rst  = new GPIO("15"); //GPIO22
                rst->export_gpio();
                rst->setdir_gpio("out");
                rst->setval_gpio("0");
                usleep(500000);
                rst->setval_gpio("1");
        //      usleep(50000);
                spi.setup(ch, 3000000);
                spi.spiMode = SPI_MODE_0;
        //      spi.spiDelay = 10000;

                uint8_t f[2];
                f[0] = (RH_RF95_REG_01_OP_MODE | RH_SPI_WRITE_MASK);
                f[1] = (RH_RF95_MODE_SLEEP | RH_RF95_LONG_RANGE_MODE);

                cs->setval_gpio("0");
                usleep(50000);
                int e = spi.readWrite(ch, f, sizeof(f));
        //      usleep(10000);
                cs->setval_gpio("1");
                usleep(10000);

                if(e) cout << "write ok: " << endl;

                uint8_t r[1];
                r[0] = RH_RF95_REG_01_OP_MODE;

                cs->setval_gpio("0");
                usleep(50000);
                e = spi.readWrite(ch, r, sizeof(r));
        //      usleep(50000);
                cs->setval_gpio("1");

                usleep(10000);
                if(e) cout << "write/read ok: " << endl;
                cout << "The Result is: " << int(r[0]) << endl;
 
                usleep(500000);
                cs->setval_gpio("0");
                rst->setval_gpio("0");
                cs->unexport_gpio();
                rst->unexport_gpio();

                close(spi.getFD(ch));
        }
};

int main( ){
        ofSetupOpenGL(1024,768, OF_WINDOW);
        ofRunApp( new ofApp());
}
```

closed 2016-12-09 by kashimAstro