kashimAstro/ofxGPIO #9: error open frameworks compiling with window

viniciolindo, 2017-02-06:
i have this error when i compile simple button example with windows

obj/linuxarmv6l/Release/src/ofApp.o:(.data+0x0): multiple definition of `OLEDFont'
obj/linuxarmv6l/Release/src/main.o:(.data+0x0): first defined here
collect2: error: ld returned 1 exit status

kashimAstro, 2017-02-06:
Hi @viniciolindo,
I tried to compile now the example-simple-led for openFrameworks, and seems to be successful.

Reported error, it seems to incriminate a double definition within the file ofApp, the basic example does not use ofApp.h and ofApp.cpp but it's all written in src/main.cpp. you confirm??

[image]

thanks for the tip..

viniciolindo, 2017-02-06:
I'm working with ofApp.cpp and ofApp.h separated. 
I test that if import only gpio.h instead ofxGPIO.h it compile without error

kashimAstro, 2017-02-06:
ok, I made a patch that solved the dual includes.

check this:
https://github.com/kashimAstro/ofxGPIO/blob/master/src/font.h#L100

let me know if this solves the problem..

thanks and good day

closed 2017-02-16 by kashimAstro