kashimAstro/ofxGPIO #15: Can't compare ofstream or ifstream to int in gpio.cpp

Taprik, 2017-10-24:
Dear KashimAstro,

I've already use your addon with success on a PI3 but this time it doesn't compile.
I use the i2c example.
I've install evrything yesterday (last Rasbian, last stable of, ...)
The errors are about ofstream or ifstream that can't be compared in the gpio.cpp
I'm giving you all the errors and hope it helps.
-------------------------------------------------------------
/home/pi/openframeworks/addons/ofxGPIO/src/gpio.cpp:28:20: error: no match for ‘operator<’ (operand types are ‘std::ofstream {aka std::basic_ofstream<char>}’ and ‘int’)
     if (exportgpio < 0){
         ~~~~~~~~~~~^~~
/home/pi/openframeworks/addons/ofxGPIO/src/gpio.cpp:28:20: note: candidate: operator<(int, int) <built-in>
/home/pi/openframeworks/addons/ofxGPIO/src/gpio.cpp:28:20: note:   no known conversion for argument 1 from ‘std::ofstream {aka std::basic_ofstream<char>}’ to ‘int’
In file included from /usr/include/c++/6/bits/stl_algobase.h:64:0,
                 from /usr/include/c++/6/bits/char_traits.h:39,
                 from /usr/include/c++/6/ios:40,
                 from /usr/include/c++/6/istream:38,
                 from /usr/include/c++/6/fstream:38,
                 from /home/pi/openframeworks/addons/ofxGPIO/src/gpio.cpp:1:
/usr/include/c++/6/bits/stl_pair.h:441:5: note: candidate: template<class _T1, class _T2> constexpr bool std::operator<(const std::pair<_T1, _T2>&, const std::pair<_T1, _T2>&)
     operator<(const pair<_T1, _T2>& __x, const pair<_T1, _T2>& __y)
     ^~~~~~~~
/usr/include/c++/6/bits/stl_pair.h:441:5: note:   template argument deduction/substitution failed:
/home/pi/openframeworks/addons/ofxGPIO/src/gpio.cpp:28:22: note:   ‘std::ofstream {aka std::basic_ofstream<char>}’ is not derived from ‘const std::pair<_T1, _T2>’
     if (exportgpio < 0){
                      ^
In file included from /usr/include/c++/6/bits/stl_algobase.h:67:0,
                 from /usr/include/c++/6/bits/char_traits.h:39,
                 from /usr/include/c++/6/ios:40,
                 from /usr/include/c++/6/istream:38,
                 from /usr/include/c++/6/fstream:38,
                 from /home/pi/openframeworks/addons/ofxGPIO/src/gpio.cpp:1:
/usr/include/c++/6/bits/stl_iterator.h:298:5: note: candidate: template<class _Iterator> bool std::operator<(const std::reverse_iterator<_Iterator>&, const std::reverse_iterator<_Iterator>&)
     operator<(const reverse_iterator<_Iterator>& __x,
     ^~~~~~~~
/usr/include/c++/6/bits/stl_iterator.h:298:5: note:   template argument deduction/substitution failed:
/home/pi/openframeworks/addons/ofxGPIO/src/gpio.cpp:28:22: note:   ‘std::ofstream {aka std::basic_ofstream<char>}’ is not derived from ‘const std::reverse_iterator<_Iterator>’
     if (exportgpio < 0){
                      ^
In file included from /usr/include/c++/6/bits/stl_algobase.h:67:0,
                 from /usr/include/c++/6/bits/char_traits.h:39,
                 from /usr/include/c++/6/ios:40,
                 from /usr/include/c++/6/istream:38,
                 from /usr/include/c++/6/fstream:38,
                 from /home/pi/openframeworks/addons/ofxGPIO/src/gpio.cpp:1:
/usr/include/c++/6/bits/stl_iterator.h:355:5: note: candidate: template<class _IteratorL, class _IteratorR> bool std::operator<(const std::reverse_iterator<_Iterator>&, const std::reverse_iterator<_IteratorR>&)
     operator<(const reverse_iterator<_IteratorL>& __x,
     ^~~~~~~~
/usr/include/c++/6/bits/stl_iterator.h:355:5: note:   template argument deduction/substitution failed:
/home/pi/openframeworks/addons/ofxGPIO/src/gpio.cpp:28:22: note:   ‘std::ofstream {aka std::basic_ofstream<char>}’ is not derived from ‘const std::reverse_iterator<_Iterator>’
     if (exportgpio < 0){
                      ^
In file included from /usr/include/c++/6/bits/stl_algobase.h:67:0,
                 from /usr/include/c++/6/bits/char_traits.h:39,
                 from /usr/include/c++/6/ios:40,
                 from /usr/include/c++/6/istream:38,
                 from /usr/include/c++/6/fstream:38,
                 from /home/pi/openframeworks/addons/ofxGPIO/src/gpio.cpp:1:
/usr/include/c++/6/bits/stl_iterator.h:1137:5: note: candidate: template<class _IteratorL, class _IteratorR> bool std::operator<(const std::move_iterator<_IteratorL>&, const std::move_iterator<_IteratorR>&)
     operator<(const move_iterator<_IteratorL>& __x,
     ^~~~~~~~
/usr/include/c++/6/bits/stl_iterator.h:1137:5: note:   template argument deduction/substitution failed:
/home/pi/openframeworks/addons/ofxGPIO/src/gpio.cpp:28:22: note:   ‘std::ofstream {aka std::basic_ofstream<char>}’ is not derived from ‘const std::move_iterator<_IteratorL>’
     if (exportgpio < 0){
                      ^
In file included from /usr/include/c++/6/bits/stl_algobase.h:67:0,
                 from /usr/include/c++/6/bits/char_traits.h:39,
                 from /usr/include/c++/6/ios:40,
                 from /usr/include/c++/6/istream:38,
                 from /usr/include/c++/6/fstream:38,
[... 542 lines cut ...]
                 from /usr/include/c++/6/bits/ios_base.h:41,
                 from /usr/include/c++/6/ios:42,
                 from /usr/include/c++/6/istream:38,
                 from /usr/include/c++/6/fstream:38,
                 from /home/pi/openframeworks/addons/ofxGPIO/src/gpio.cpp:1:
/usr/include/c++/6/bits/basic_string.h:5175:5: note: candidate: template<class _CharT, class _Traits, class _Alloc> bool std::operator<(const _CharT*, const std::__cxx11::basic_string<_CharT, _Traits, _Alloc>&)
     operator<(const _CharT* __lhs,
     ^~~~~~~~
/usr/include/c++/6/bits/basic_string.h:5175:5: note:   template argument deduction/substitution failed:
/home/pi/openframeworks/addons/ofxGPIO/src/gpio.cpp:86:22: note:   mismatched types ‘const _CharT*’ and ‘std::basic_ifstream<char>’
     if (getvalgpio < 0){
                      ^
In file included from /usr/include/c++/6/bits/ios_base.h:46:0,
                 from /usr/include/c++/6/ios:42,
                 from /usr/include/c++/6/istream:38,
                 from /usr/include/c++/6/fstream:38,
                 from /home/pi/openframeworks/addons/ofxGPIO/src/gpio.cpp:1:
/usr/include/c++/6/system_error:200:3: note: candidate: bool std::operator<(const std::error_code&, const std::error_code&)
   operator<(const error_code& __lhs, const error_code& __rhs) noexcept
   ^~~~~~~~
/usr/include/c++/6/system_error:200:3: note:   no known conversion for argument 1 from ‘std::ifstream {aka std::basic_ifstream<char>}’ to ‘const std::error_code&’
/usr/include/c++/6/system_error:274:3: note: candidate: bool std::operator<(const std::error_condition&, const std::error_condition&)
   operator<(const error_condition& __lhs,
   ^~~~~~~~
/usr/include/c++/6/system_error:274:3: note:   no known conversion for argument 1 from ‘std::ifstream {aka std::basic_ifstream<char>}’ to ‘const std::error_condition&’

kashimAstro, 2017-10-24:
Hi @Taprik,
thanks for the tip! 
can you tell me which openframeworks release you are using?
and pass me a: uname -a of your raspbian with a g++ --version..

thanks and good day!

Taprik, 2017-10-24:
Hi KashimAstro,
Thanks for your prompt reply.
I've switch to Jessie and now it's working fine.
I was using Stretch and it's definitly not working
https://forum.openframeworks.cc/t/rpi-install-dependency-error-recompiling-poco/28013/4
I was using of 0.9.8.
Don't remenber the g++ version.
Best regards

kashimAstro, 2017-10-24:
I have to make some changes to support Debian Stretch.
Leave this issue open, I'll give you updates over a few days!

kashimAstro, 2017-12-28:
solved this, check new update.

cheers

closed 2017-12-28 by kashimAstro