kashimAstro/ofxGPIO #16: example-apa102-ledstrip not lighting leds with Stretch

chilina, 2017-12-21:
The example compiles, and when it runs a terminal message indicates the SPI setup was successful.  This happens whether the example is run with sudo or without.  The example lights up the leds with Jessie and the same hardware.

kashimAstro, 2017-12-22:
Hello, thanks for the report.

right now i'm traveling for work i do not have the tools with me for checks.
I will update you as soon as I go back to Italy at the beginning of January!

you can give me these info:
uname -a
(in case you're using openframeworks)
openframeworks version

chilina, 2017-12-22:
Hi Dario,
Thanks for such a quick response and I'm not in a rush to move to Stretch.  I'll try to do some work on this too over the next few weeks.
uname -a:
Linux statler 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux
I am using the "master branch" of oF, which I cloned from https://github.com/openFrameworks/openFrameworks.git with modifications according to Christopher Baker's post https://forum.openframeworks.cc/t/compiling-of-in-raspbian-stretch/27562/15.  I have oF folders for Method 1 and Method 2.

kashimAstro, 2017-12-27:
Hi @chilina,

the BUG apa102 is theoretically resolved.
tested on debian stretch. 

if you can test on jessie it would be nice.

cheers

chilina, 2017-12-27:
Wow Dario thanks!  I'll test it on Jessie soon and post the results here.  What did you do to resolve it with Stretch?

closed 2017-12-27 by chilina

kashimAstro, 2017-12-28:
to solve the problem apa102 just add a memset in the
readWrite method of spi2.h, view here:
https://github.com/kashimAstro/ofxGPIO/blob/master/src/spi2.h#L32

however, in these last days I have made some changes and additions to ofxGPIO,
with changes also on LED class that manages apa102 
i advise you to rediscover the master and take a look at it.

has been added (if you work without openframeworks):
Gtk+ UI a simple bridge for gtk ui interface
TCP Unix socket for communication between application or more raspberry 

and added:
simplified syntax for class GPIO and MCP
check readme and documentation.

and read these examples:

> example-mcp-simple-syntax
> example-simple-syntax-gpio

> 
(TCP server, control pin: GPIO17)
> example-server-led

(TCP client, ON/OFF button example for control: example-server-led )
> example-gtk2-client

(Gtk+ UI sample)
> example-gtk2-led
> example-gtk2-ui

(this example work only openframeworsk, 
 used to control I/O of the whole GPIO through a graphical interface.)

> example-gpio-pindescription