kashimAstro/ofxGPIO #18: Use SPI1 for APA102

yiank, 2018-03-11:
Hello,
Can I use SPI1 and pins 20,21 for driving APA102?
I tried to change the port on setupAPA102 but didn't worked because there is a bug in the led.h at the setAPA102 (spi.readWrite selects always 0)
`spi.readWrite(0, (unsigned char*)buffer0, 1)`

But even after change this to 1, still cannot use pin 20 and 21.

Do you have any recommendation? 

thank you

kashimAstro, 2018-03-12:
Hi, thanks for this, I just fixed the bug.

for enable SPI1 read this:
http://terminal28.blogspot.it/2016/05/enabling-spi1-on-raspberry-pi-bzero23.html

if you use a raspberry pi 3 with active bluetooth
you have to disable it, to make SPI1 work.

yiank, 2018-03-12:
Thank you for the reply.

I manage to used the SPI1 following your advice but I have also to edit the spi2.h. In this header you have restrict the ports to use only SPI0 with this code:
`class SPI2 { 
    public:
	string      spiDev0 = "/dev/spidev0.0" ;
	string      spiDev1 = "/dev/spidev0.1" ;` 

To solve it quick and dirty, I just edited to this:
`string      spiDev0 = "/dev/spidev1.0" ;`

and I called the setupAPA102 with the channel to 0:
`apa.setupAPA102(0);`

Anyway it works for me now, thank you!

kashimAstro, 2018-03-12:
later i fix this too, thank you!

kashimAstro, 2018-03-13:
I added a second setup method in the SPI class,
where you can pass the device as the first parameter.

and i added a path parameter to the setupAPA102(path,channel,speed) method
with default on SPI0.

unfortunately I can not do tests at the moment because I'm traveling,
if by chance you can do a test, so I close this issues..

Ciao

yiank, 2018-03-13:
Yes, it's working! 
Thank you for the fast support. Your addon gives to the openFrameworks/Rpi new potential.

closed 2018-03-13 by yiank