kashimAstro/ofxGPIO #19: 12c examples

fred-dev, 2018-04-17:
Hi, there are some errors in the 12C examples, the OF version is actually without OF (in the makefile) and does not run.

I am trying to use an LIS3DH adafruit accelerometer, but I want to check the connection first...

fred-dev, 2018-04-18:
Hey, I get this:
Linux raspberrypi 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv7l GNU/Linux
I am also trying to get SPI for an MCP3008, I don't manage to get any input (I have used the chip before). In the meantime can you recommend a version of OF and the stretch/or previous OS for linux that will function?

kashimAstro, 2018-04-18:
Hi @fred-dev , 

I just fixed the examples for the i2c bus scanner, now you should see the addresses on the bus!

the example for the openframeworks scanner wired the path to the device: /dev/i2c-1 in the code
you can modify it as you wish.

regarding SPI there should not be any problem, both the example i2c and SPI must be executed with a user who can access you device, you can check if your user belongs to the i2c group and spi with command: id or run the example with sudo.

you can also refer to this issues:
https://github.com/kashimAstro/ofxGPIO/issues/18

for some additions made to the SPI class. 
make sure the MCP example points to the right path for the SPI device you are using.

try to look at this example that takes the path of the spi device as a parameter in the setup method
https://github.com/kashimAstro/ofxGPIO/blob/master/example-mcp-simple-syntax/src/main.cpp

please let me know
Greetings

fred-dev, 2018-04-19:
Hi, thanks, I have no idea what I was doing wrong but I started from scratch and the MPC worked fine, I had used it a few months ago so I had already found thee quirks, just wanted to do a poc before I send a PCB for fabrication.

The i2c is a different issue. I can confirm everything is working, but just not with your code. I have added a picture that I can scan the 12c bus with a command, but your addon does not work, I have been using elevated permissions. Knowing that it can work I will go ahead with my PCB, needless to say it would be great to be able to use your code and have everything in a single app, rather than run the 12c from another script.

Cheers

Fred

PS, thanks a million for this work, ofxGpio is really extensive and well put together, you are making a lot of crazy projects happen!!

From: Dario Longobardi
Sent: Wednesday, 18 April 2018 7:44 PM
To: kashimAstro/ofxGPIO
Cc: Frederick Rodrigues; Mention
Subject: Re: [kashimAstro/ofxGPIO] 12c examples (#19)

Hi @fred-dev ,
I just fixed the examples for the i2c bus scanner, now you should see the addresses on the bus!
the example for the openframeworks scanner wired the path to the device: /dev/i2c-1 in the code
you can modify it as you wish.
regarding SPI there should not be any problem, both the example i2c and SPI must be executed with a user who can access you device, you can check if your user belongs to the i2c group and spi with command: id or run the example with sudo.
you can also refer to this issues:
#18
for some additions made to the SPI class.
make sure the MCP example points to the right path for the SPI device you are using.
try to look at this example that takes the path of the spi device as a parameter in the setup method
https://github.com/kashimAstro/ofxGPIO/blob/master/example-mcp-simple-syntax/src/main.cpp
please let me know
Greetings

kashimAstro, 2018-04-19:
Hi, thanks for feedback!

unfortunately, I can only reach a remote raspberry that is in my office in Italy, while I am now in switzerland. I'd like to solve this, can you help me debug?

if you can! add a: `cout << result << endl;` to the scanner example, immediately after this line:
https://github.com/kashimAstro/ofxGPIO/blob/master/example-i2c-scanner/src/main.cpp#L28
and pass me the output.

Grazie!

fred-dev, 2018-04-19:
> src/main.cpp#L28

It gives me:

65415

Cheers - and I am ok for a few days, thanks so much for the effort.

kashimAstro, 2018-04-19:
the whole list contains only the value: 65415? 
Is the chip conn this is strange smbus should return 0 or 255 if it finds an address..
ected when you run example?

fred-dev, 2018-04-28:
Ok, I had a chance to try this again, I was wrong it is not only 6514.

The 24th value is read as 31. My 12c device is addressed at 18, as shown in the previous picture.

Could it be that this value is now coming in as octal and with an offset?

My sensor should be 18, but if there is now a start address offset (it starts at 1 not zero) and the values are octal.

031 octal is 19 hex and 25 decimal?

kashimAstro, 2018-04-29:
Hi @fred-dev  thanks for this,
in the next few days I solve this problem!