kashimAstro/ofxGPIO #20: Gpio io

fred-dev, 2018-05-11:
I have a strange issue, I am running my app with elevated (Sudo) permissions. I don't get any response from a simple push button until I run node red to test the setup. Then my OF app will run perfectly until a reboot. 

The only obvious difference I see is node red has a checkbox for setting pull up and pull down states for input pins. Is this function available in ofxgpio or is there something deeper at play.

I am using the latest OF and stretch install.

kashimAstro, 2018-05-13:
Hi @fred-dev sorry for delay, but i'm traveling for work.

you can set the status (UP / DOWN) of the pin during setup, the last parameter accepts:
LOW or HIGH.

example:
pin.setup(GPIO17, OUT, LOW);

reference:
https://github.com/kashimAstro/ofxGPIO/blob/master/src/gpio.h#L37
https://github.com/kashimAstro/ofxGPIO/blob/master/src/gpio.h#L42

sorry, what is node red?

fred-dev, 2018-05-14:
Sorry for not explaining, node red is a utility that comes with the
raspberry pi raspbian installation that can also access the gpio pins
https://nodered.org.

I will try setting the low/high option as soon as I get back to my project
next week.

Cheers

fred-dev, 2018-05-22:
After some more serious debugging, I still have this error. Initially using code from your examples I get no response from ofxGpio using an input pin. If I use python or node-red to check the gpio (using an input as I am using a button), it works, once I have done that I can then use code with ofxGpio. This works for some time and then stops. I can get the button working again 100 percent of the time by udsing node red or python to acceess the gpio and then it will work perfectly. ofxGpio stops working 100% of the time if the pi loses power, other . It seems it does not initialise the input pin properly (output pins always work).

Also your above comment to use gave me a more often functioning result, you suggest this:

pin.setup(GPIO17, OUT, LOW);

is a bit confusing, in your examples you use the syntax

```
gpio6   = new GPIO("6");
			gpio21  = new GPIO("21");
			gpio21->export_gpio();
                        gpio21->setdir_gpio("in");
```

How should this work? Maybe this is what is giving me strange results.

                        ```
gpio21  = new GPIO("21", "in", "high");
			gpio21->export_gpio();
                        gpio21->setup("21", "in" , high);

```
It is not clear from your code and as far as I can see your examples use a syntax that in my code does not work for a simple input, when my circuit works perfectly with other code. I would really like this project to have a button....

kashimAstro, 2018-05-23:
Hi @fred-dev,

This is very strange, i have a lot of work around on raspberry using I / O
and i've never encountered this problem. 

I can tell you with certainty that ofxgpio unlike other library
works with: "/sys/class/gpio/" for the gpio settings, and does not work on "/dev/mem"

if you are having problems with I/O through ofxgpio you can try to execute your own execs to initialize the pins and manage I/O

example on pin 17:
```
/* create pin 17 */
echo 17 > /sys/class/gpio/export 

/* conf pin direction (in/out) */
echo out > /sys/class/gpio/gpio17/direction

/* write value ON */
echo 1 > /sys/class/gpio/gpio17/value

/* write value OFF */
echo 0 > /sys/class/gpio/gpio17/value

/* delete GPIO17 */
echo 17 > /sys/class/gpio/unexport

```

this weekend I will a check on ofxgpio to verify this your report. I keep you updated!

Thanks!
Dario

CyrCom, 2018-08-21:
Hello @kashimAstro ,
Thanks for the addon. I also want to change the pull up / pull down resistor that the bcm chip provides.
I don't know a lot, but I think this is very different from the output value you can set ?
I have a button on pin 4, connected to ground on the other side:
```
gpio4.setup("4");
gpio4.export_gpio();
```
works great, because, pin4 (BCM) has by default PULL UP resistor.

Well now I need a second button and there is no more defaulted pin like this one, so I need to be able to set pin17 to a pull-up resistor. Is this provided with the addon ??

Thanks :)