kashimAstro/ofxGPIO #22: Cannot make SPI work

Meach, 2018-07-06:
Hello,

I am trying to use the SPI with MCP23S08 (IO Expander).
I tried with Arduino before to make sure my code works and I read/write the correct registers but I cannot make it work with the RPI.

I connected the chip as follow:
- MOSI pin 19
- MISO pin 21
- SCLK pin 23
- CS pin 24

I also power the chip with +3.3V (pin 1)

Here is my code, can you spot something I am doing wrong?

```
SPI2 _spi;

//--------------------------------------------------------------
void ofApp::setup(){
	// Setup our chip on SPI 0
	_spi.setup("/dev/spidev0.0", 0, 1000000);
	
	// Configure chip: set all GPIO as input
	char data[3];
	data[0] = 0x40;		// Opcode
	data[1] = 0x00;		// IODIR - I/O Direction register
	data[2] = 0xff;		// all GPIO pins are inputs

	// Write the 3 data bits on channel 0
	_spi.readWrite(0, (unsigned char *)data, 3);
}

//--------------------------------------------------------------
void ofApp::update(){
	// Read all 8 IO expander inputs
	char data[3];
	data[0] = 0x41;		// Opcode
	data[1] = 0x09;		// GPIO - I/O port register address
	data[2] = 0x00;		// Dummy to read the value

	// Write the 3 data bits on channel 0
	val = _spi.readWrite(0, (unsigned char *)data, 3);
	
	ofLogVerbose() << "Val = " << val;
}
```

kashimAstro, 2018-07-09:
Hi @Meach, sorry for delay.

right now i'm traveling for work, return to the lab this weekend, i keep you updated!

Ciao
Dario

Meach, 2018-07-10:
Hi @kashimAstro,
Thanks for answer. I am using wiringPi now instead and managed to make it work. Also looking at the wiringPi documentation, I believe my problem is here:

// Write the 3 data bits on channel 0
val = _spi.readWrite(0, (unsigned char *)data, 3);

I expect the value read to be returned with readWrite function, where data variable might be overwritten with the read data.