kashimAstro/ofxGPIO #24: Setting pull up resistors

DanielBlackburn, 2018-10-05:
I can't find out how to set the pull up resistors in ofxGPIO? or is this feature not supported.

kashimAstro, 2018-10-05:
you can choose whether the PIN is HIGH or LOW
with the third parameter of the setup method of the GPIO class.

Example:
`gpio.setup(GPIO17,OUT,LOW);`

or you can choose to apply a hardware pull-up.
Example:
[image]

DanielBlackburn, 2018-10-05:
Thanks for the quick response. I thought in the top example though that the third parameter sets the pin LOW, not the pins pull up resistor.

If that is the case would the following code set the pin to input and set the pull up resistor to high?
button.setup(GPIO17,IN,HIGH);

DanielBlackburn, 2018-10-05:
Thanks for the second example but I don't want to use hardware pull-ups. I want to use the Pi's internal pull up resistors.

DanielBlackburn, 2018-10-05:
Wiring Pi has a separate function (pullUpDnControl (int pin, int pud) )to set the pins internal resistors to either off, up, or ground. This is set independent of the pins direction and whether it is high or low. I am looking for the equivalent of this in ofxGPIO.

kashimAstro, 2018-10-05:
sorry but I thought you were referring to the initial status of the PIN if you talk about the PUD status, the gpio class at the moment is not able to manage it.. ofxGPIO include also BCM2835 you can try starting from:

https://github.com/kashimAstro/ofxGPIO/blob/master/src/bcm2835.c#L409-L423

however, I also recommend taking a look at wiringpi
which implements these features in a simpler way.

good day!

kashimAstro, 2018-10-05:
just for information, ofxGPIO was born with the purpose of managing multiple types of board (not just raspberry), many of which do not provide a pinout with integrated resistors.

that's why I decided to manage I/O with sysfs:
https://www.kernel.org/doc/Documentation/gpio/sysfs.txt

however this is not always comfortable, as soon as I'm free from work I'll try to implement new boards-specific features.

DanielBlackburn, 2018-10-05:
OK, I understand. Thanks for the clarification. For now I'll use system() to call wiring pi at runtime.

I spotted this in bcm2835.h : 
```
/// Sets the Pull-up/down mode for the specified pin. This is more convenient than
    /// clocking the mode in with bcm2835_gpio_pud() and bcm2835_gpio_pudclk().
    /// \param[in] pin GPIO number, or one of RPI_GPIO_P1_* from \ref RPiGPIOPin.
    /// \param[in] pud The desired Pull-up/down mode. One of BCM2835_GPIO_PUD_* from bcm2835PUDControl
    extern void bcm2835_gpio_set_pud(uint8_t pin, uint8_t pud);
```
If I get time I'll also look at exposing this functionality on the Pi at least.

Thanks very much for all your work on ofxGPIO.