kashimAstro/ofxGPIO #33: Adapting read-write example to distance sensor

AnasGhrab, 2019-06-26:
I'm trying to adapt the read-write example to a distance sensor (HC-SR04), sending a Trigger and getting the Echo result. I'd be glad if I get help with this not working code :

```
#include "ofMain.h"
#include "ofAppNoWindow.h"
#include "ofxGPIO.h"

class noWin : public ofBaseApp{
        public:
		GPIO* gpio17;
                 GPIO* gpio39;
                 GPIO* gpio40;

		string state_button, echo;

   void setup(){
      gpio17  = new GPIO("17");
      gpio39  = new GPIO("39"); // Trigger (out)
      gpio40  = new GPIO("40"); // Echo (in)

      gpio17->export_gpio();
      gpio39->export_gpio();
      gpio40->export_gpio();

      gpio17->setdir_gpio("in"); // State
      gpio39->setdir_gpio("out"); // Trigger
      gpio40->setdir_gpio("in"); // Echo

		}

void update(){
      if(state_button == "1"){
          gpio39->setval_gpio("1");
          usleep(10);
          gpio40->getval_gpio(echo);
          ofLog()<< echo;
          }
      if(state_button == "0"){
          gpio39->setval_gpio("0");
          usleep(5000);
          }
       }

   void exit(){
      gpio17->unexport_gpio();
      gpio39->unexport_gpio();
      gpio40->unexport_gpio();
     }
};

int main( ){
  ofAppNoWindow window;
	ofSetupOpenGL(&window, 0,0, OF_WINDOW);
	ofRunApp( new noWin() );
}
```

kashimAstro, 2019-06-27:
There are two things wrong with this example:
1) the numbering of triggers and echoes is wrong
there are no pins numbered with 39 and 40
2) state_button is not set.

try starting with this example (untested):
https://gist.github.com/kashimAstro/41662bd8d78246efb1820a210c7c3e60

there are two variables in the class no_window:
int trigger = 0;  <--- number pin trigger 
int eco     = 0;  <--- number pin echo 

if you are not sure of the numbering turn me a picture of your scheme
and remember to use resistors on the echo pin.

kashimAstro, 2019-06-27:
[image]

in this scheme the nmber of trigger is GPIO18 and echo GPIO23.

AnasGhrab, 2019-06-27:
Thank you very much ! I'll try it and keep you informed.

AnasGhrab, 2019-06-27:
I'm getting this error : 

```
obj/linuxarmv6l/Release/src/main.o: In function `no_window::~no_window()':
main.cpp:(.text._ZN9no_windowD2Ev[_ZN9no_windowD5Ev]+0x28): undefined reference to `HC_SR04::~HC_SR04()'
obj/linuxarmv6l/Release/src/main.o: In function `no_window::~no_window()':
main.cpp:(.text._ZN9no_windowD0Ev[_ZN9no_windowD5Ev]+0x28): undefined reference to `HC_SR04::~HC_SR04()'
obj/linuxarmv6l/Release/src/main.o: In function `main':
main.cpp:(.text.startup+0x118): undefined reference to `HC_SR04::HC_SR04()'
collect2: error: ld returned 1 exit status
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:403: recipe for target 'bin/test-HC-SR04' failed
make[1]: *** [bin/test-HC-SR04] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/test-HC-SR04'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:124: recipe for target 'Release' failed
make: *** [Release] Error 2
```

AnasGhrab, 2019-06-27:
I had to comment   ```HC_SR04();``` and  ```~HC_SR04();``` in the HC_SR04 class. But it is still not working. The Raspberry freezes.